DIT112-V20/group-09 #67: Upgrade to GCC10 for better C++2a support

AeroStun, 2020-05-15:
See `<compare>` and `<span>`

platisd, 2020-05-20:
I was trying to use `std::span` with GCC 9.2 and it wouldn't be there. Good to know it came with GCC 10. I was waiting for it! :tada:

AeroStun, 2020-05-20:
> https://en.cppreference.com/w/cpp/compiler_support

GNU libstdc++ with GCC10, LLVM libc++ with Clang/LLVM7, MS STL with MSVC19.26, AppleClang libc++ with XCode10

Technically we already use the pre-c++2a `gsl::span`, so that transition will be seamless for us

platisd, 2020-05-20:
Were you using [Microsoft's implementation](https://github.com/Microsoft/GSL) for GSL?

AeroStun, 2020-05-20:
Correct; however I mostly included the GSL for the its general purpose: a support library for the CppCoreGuidelines
If you're looking for a portable way to have spans in any ISO C++ version, I can recommend [span_lite](https://github.com/martinmoene/span-lite)

platisd, 2020-05-20:
> Correct; however I mostly included the GSL for the its general purpose: a support library for the CppCoreGuidelines
> If you're looking for a portable way to have spans in any ISO C++ version, I can recommend [span_lite](https://github.com/martinmoene/span-lite)

That's exactly what I was after, thanks! :+1: