DIT112-V20/group-09 #71: GCC somehow invalidates `board_data` on cross-threads access

AeroStun, 2020-05-17:
Direct cause of [this CI failure](https://github.com/DIT112-V20/group-09/pull/69/checks?check_run_id=682619954)

AeroStun, 2020-05-17:
Update: within a single stacktrace, two frames accessing the same address obtain different results
[image]
[image]

AeroStun, 2020-05-17:
Only possible explanation is that the DLL somehow ends up with a different virtual memory space, but this is supposedly impossible since previous demos have demonstrated that user-code actions can be observed by the main thread

AeroStun, 2020-05-17:
Better yet, with Clang: from a single stack frame, depending on how the pointer to the object is accessed, different results are obtained (see that from the lambda I get a `pin_modes` of size 1, and from the global, which point at the same address, I get a `pin_modes` of size 0)
[image]

AeroStun, 2020-05-17:
Update: tried setting an x86 hardware watchpoint using LLDB on `board_info->pins_caps._M_impl._M_start` (yes I use libstdc++), and the watchpoint does not fire

AeroStun, 2020-05-17:
We have reached a whole other level
[image]

AeroStun, 2020-05-17:
It only gets better...
[image]

AeroStun, 2020-05-18:
Have I mentionned that debuggers are useless?
[image]

platisd, 2020-05-18:
This seems like the definition of an undefined behavior. Is there something that's not initialized perhaps?

AeroStun, 2020-05-18:
Undefined behavior does not affect operator `sizeof` since its operands are unevaluated (only the type of the expression is used)
I suspect that LLDB is trying to JIT some of its evaluations with different build options

AeroStun, 2020-05-18:
Next assumption: broken ABI

AeroStun, 2020-05-18:
CONFIRMED: broken ABI
- SMCE and its tests compile in non-debug mode, hence they use `std::__cxx1998::vector`
- ArduinoRuntime compiles in debug mode, hence it uses the 56 bytes long (why the hell do they use that extra space for? Gotta go read their debug impl now) `std::__debug::vector`

AeroStun, 2020-05-18:
Morale of the story: too much debugging kills the debugging
And second morale is that debuggers can't be trusted since they don't tell the user how they got the results from expression evaluation; only Valgrind stands trustworthy

closed 2020-05-18 by AeroStun